quince-science/QuinCe #108: Unrecognised run types are ignored

squaregoldfish, 2016-01-27:
I think they should be exceptions.

squaregoldfish, 2017-10-29:
They should also be checked at file upload time, so the file is rejected immediately

jonasfh, 2018-02-12:
Discussing this with Rocio, we decided that the user will need to revisit the instruments - page to add a new run-type there.

squaregoldfish, 2018-02-12:
What's the reasoning?

jonasfh, 2018-02-13:
There are several reasons: 
- Rocio thinks it is good for the scientists to have to consider their decisions, not just press IGNORE
- I like the idea because it is much easier to implement
- And: most possibly this don't happen to often, so it will not be a big problem to have to do it this way
- I initially asked Rocio her opinion, as I did not see where to logically add the UI parts for this

closed 2018-03-01 by jonasfh