quince-science/QuinCe #260: MySQL time zone

squaregoldfish, 2016-09-28:
At the moment, timestamps are stored in the time zone of the server on which MySQL is running, and converted to UTC on the way out. This is fine as long as the server's time zone never changes, but it will for daylight savings time.

All times should be stored in the database as 'raw' UTC.

squaregoldfish, 2016-09-28:
1. Set the MySQL server to UTC on startup
2. Replace all getConnection instances with a special method that sets the timezone to UTC.
3. Ensure that all dates are being stored in UTC (probably don't need to do anything explicit - it should just happen).
4. Convert all dates to UTC on the way out (already done).

squaregoldfish, 2016-11-22:
https://dev.mysql.com/doc/refman/5.5/en/datetime.html

This implies that we should use DATETIME in the database instead of TIMESTAMP, because TIMESTAMP tries to do conversion to UTC and DATETIME stores the raw value that's passed in. Need to read up on the implications for the above plan

karlmattsmith, 2016-11-22:
FYI.  In SOCAT, neither are used due to this timezone issue.  Instead, a UNIX time value (sec since 1970-01-01 00:00:00 UTC - or similar) is stored.  This value can be displayed (for debugging or whatever) as a date/time in SQL using 'from_unixtime'.  In earlier versions of SOCAT when data was being stored in MySQL, the individual year/month/day/hour/minute/second values were being store as well.

squaregoldfish, 2016-11-22:
Ah, a good approach to sidestep the whole issue! I'll give it some thought. It's probably not any more intensive processing-wise either. Thanks for the tip!

squaregoldfish, 2017-02-28:
Convert everything to JSR310 classes while we're at it

squaregoldfish, 2018-01-22:
Done

closed 2018-01-22 by squaregoldfish