quince-science/QuinCe #268: Correcting SST/Eq

squaregoldfish, 2016-10-19:


squaregoldfish, 2017-09-19:
Do we have to account for an offset between the equilibrator and the CO2 sensor? Is this even possible, given the data we have available? I have a suspicion that it isn't. Also, the equilibrator and the CO2 sensor should be at roughly the same temperature, so there probably isn't a significant change anyway. Check with Denis though.

squaregoldfish, 2018-04-25:
Once the correct offset has been calculated, the equilibrator temperature, pressure and CO2 measurements should be moved to match the SST record.

We need to record somewhere in the dataset what offset was applied, so it can be recorded in the metadata when the data set is published. In the long run there are a number of things like this we need to record (what decisions QuinCe made while processing the data), so we'll need a mechanism to handle it.

squaregoldfish, 2018-05-23:
Having discussed this with Denis, this probably shouldn't be automated, because it's too prone to error. We can either have a fixed offset defined as part of the instrument, or make a UI for the user to adjust the offset for the data set when it's defined (after data extraction but before calculation).

The first option is simplest, but assumes that the offset won't change over time. And since it's sensitive to flow rate, I'm not sure that's a valid assumption.

jonasfh, 2018-05-30:
There are some approaches to make this practical: 

1. Split the dataset-data table into two tables: one handling the position (incl. time) and one handling the parameter data. This table will be huge (we've discussed this)
2. Split dataset-data into three tables: One handling the position (incl. time), one containing parameters at the intake, one with parameters at the equilibrator.

Both these approaches allow for attaching data to a different time and location information than what is in the collected data. Approach one is the most generic and flexible, but also the least efficient. Using the second approach, the db-routine rewriting will be minimal.

squaregoldfish, 2018-06-28:
Denis's approach is to use interpolation - so we would take the SST value, and interpolate the measurements on the equilibrator side of things to get a value at the correct offset from the SST measurement time.

This is relatively difficult to implement with the current setup, and also the current practice of applying a QC flag to a whole record is problematic in terms of selecting which record is actually flagged. Therefore this will wait until individual sensor QC is in place (#576). This will allow individual sensor measurements to be flagged bad, and the interpolation can operate on the remaining values. 

There should be limits on the time over which interpolations can be performed before the values become invalid. In those cases there are a few options - use the closest available measurement, extrapolate the trend from the last few measurements (problematic with sensor noise and coastal high variability), or simply not calculate a value at all.

squaregoldfish, 2019-07-21:
Possible graphical interface for defining the offsets
[image]

Note that the time offsets (in purple) can show both the interval to previous sensor and total from water intake.

squaregoldfish, 2019-10-01:
**Notes from call with Denis**

Offsets cannot be fixed on an instrument, so the above idea is toast. Instead, when a dataset is defined we have to specify the offset. Only SST/EqT is important here, and we move all other values to the SST value.

I can have an automatic algorithm that tries to find the best offset. This can simply offset the two variables at several lags (up to 10 min) and work out which has the highest correlation - we choose that one. The user can then tweak it if they wish. Possible interface below:
[image]
As the user tweaks the offset, the plots update.

Denis wondered if there's a one-shot way to do this? Plotting EqT vs SST and running a least squares fit would give an optimal slope with r², but how is that translated into an offset?

squaregoldfish, 2019-10-14:
This needs to happen between the Data Extraction Job and the Locate Measurements Job. We need a way to interrupt the job flow and have a User Action stage which will bring up the interface. Once it's set, resume the job flow. Before AutoQC job is probably best, because then it won't interfere with recalculation during manual QC.

This can probably be defined at the InstrumentVariable level, and needs the DataSet status to have more options.

The chosen offset must be set as an attribute of the dataset. The Data Reduction Job must take this into account when finding values for measurements. (Base measurements off SST instead of CO2 in Locate Measurements? Or in Data Reduction work out the SST and edit the time of the measurement to match it?)

squaregoldfish, 2019-10-15:
Cross correlation is the thing we need.

Need to work out how to handle SST and EqT in different files and on different timing schemes. Maybe there's a library out there that can handle it.

squaregoldfish, 2019-11-28:
Do the offset as part of the `LocateMeasurementsJob`. Then the user can edit at the end, which will trigger recalculation from that job. (NRT will not have the option)

The offset will be stored in the dataset somehow. If it's null, the job will calculate it. Otherwise it will apply what's there.

squaregoldfish, 2020-06-12:
This can potentially be applied to any pair of sensors, and some will be negatively correlated. So the cross-correlation should find the closest to either 1 or -1.

squaregoldfish, 2020-11-03:
It may be that the offsets aren't fixed for the full length of a dataset. Do some experiments with SOCAT and ask Harald.

squaregoldfish, 2021-08-10:
Implementation plan:

- Create groups of sensors in the instrument definition. Define the linking sensors
- Algorithm for detecting peaks/troughs and lining them up through the course of a dataset
- UI for reviewing and editing match points

Offsets stored as part of the dataset properties

squaregoldfish, 2022-01-28:
Feeback from Tobias:

- Add the ability to view another plot (eg water flow) to find points where it's likely an offset change will occur
- Add a plot the shows the delta between the two variables to see where the relationships change suddenly (toggle applied offset option?). This can be alternated with the current offset time series in the bottom left
- Toggle axis locks, so we can switch between having the two variables on separate axes on the same axis. Tobias got confused by SST looking higher than EqT.
- Add correlation stats (applied and non-applied) to show whether it's getting better or not.

squaregoldfish, 2022-01-28:
It's possible that there might be a group between entries of interest, i.e.

SST <-> SSS <-> EqT

Can we make this work? I can't immediately see how.

squaregoldfish, 2022-02-24:
We seem to have something that works. Check with interested parties (Tobias/KPH/Denis)

squaregoldfish, 2022-03-16:
Handle Type columns. If we offset directly onto a measured value, it should be marked M.

This may mean that we need to store the Type in the MeasurementValues JSON.

closed 2022-05-06 by squaregoldfish