quince-science/QuinCe #496: Rationalise reference assignments into single table

squaregoldfish, 2017-08-25:
The reference assignments (gas standards and calibration coefficients and eventually externals calibrations #495) are currently stored in separate tables. This means we can't write safe generic code to interrogate these, because the table name always needs to be specified which is a security issue.

Reference assignments should all be stored in a single table, with a structure something like:
* instrument id
* reference type (gas standard, coefficients)
* reference name (name of standard/sensor)
* reference date
* values (a comma-separated list of Doubles)

squaregoldfish, 2017-08-25:
I believe the abstract `ExistingReferenceAssignmentValidator` can be restricted to always restrict the search to a given instrument, and get rid of the `restrictionField` parameters.

squaregoldfish, 2017-08-25:
Pay particular attention to the `DATETIME` requirement in #495 .

jonasfh, 2018-01-30:
Ref #597

squaregoldfish, 2022-04-08:
This has been done as part of other changes.

closed 2022-04-08 by squaregoldfish