quince-science/QuinCe #588: New 'averaging' method - last of run

squaregoldfish, 2017-11-24:
Some instruments run for a while to equilibrate, and then the last record of each run is used for the CO2 value.

We still need to be able to review the measurements running up to that point, to make sure it's equilibrating properly.

squaregoldfish, 2018-02-09:
Another option is the ability to average over the last n measurements, or the last n minutes of each run

squaregoldfish, 2018-08-01:
Regarding the last point: Do we average the sensors then do the calculation, or do the calculation and average the fCO2 values? I guess the former so we can include the SST/Sal etc in SOCAT correctly, but double-check with PIs. (Or SOCAT, so see what we *should* be doing).

MarenKK, 2020-09-22:
#408

squaregoldfish, 2024-02-13:
This has been covered by CONTINUOUS/PERIODIC measurement modes.

closed 2024-02-13 by squaregoldfish