quince-science/QuinCe #623: Error navigating to Files - tab with no instrument defined

jonasfh, 2018-01-08:
When no instrument is defined, you get an error going to the Files - tab. The Files - tab should probably be hidden if no instrument is defined.

Error output: 

> javax.el.ELException: Error reading 'listFiles' on type uk.ac.exeter.QuinCe.web.files.DataFilesBean
> 
> viewId=/data_file/file_list.xhtml
> location=/home/jhe052/eclipse-workspace/.metadata/.plugins/org.eclipse.wst.server.core/tmp1/wtpwebapps/QuinCe/data_file/file_list.xhtml
> phaseId=RENDER_RESPONSE(6)
> 
> Caused by:
> uk.ac.exeter.QuinCe.utils.MissingParamException - instrumentID parameter is null
> at uk.ac.exeter.QuinCe.utils.MissingParam.checkPositive(MissingParam.java:63)
> 
> /data_file/file_list.xhtml at line 24 and column 134 value="#{dataFilesBean.listFiles}"

squaregoldfish, 2018-01-09:
The same will apply to the data sets tab. I guess we should detect the issue and put up a message saying that an instrument needs to be defined.

squaregoldfish, 2018-01-30:
ref #669

closed 2018-01-31 by squaregoldfish