quince-science/QuinCe #629: Error clicking View Deployments if no deployments exist

jonasfh, 2018-01-16:
At the Instruments - tab your instruments are listed. For each instruments there is a View deployments - button. However, if no deployments have yet been registered, when you click View deployments you'll get an error, like: 

> Internal Error
> Something horrible has happened. Please accept my apologies. Details of the horrible thing are below. If the horrible thing keeps happening, please copy the details into an email and send it to quince@uib.no.
> uk.ac.exeter.QuinCe.utils.RecordNotFoundException: No external standard names found for instrument 21 at uk.ac.exeter.QuinCe.data.Instrument.Calibration.ExternalStandardDB.getTargets(ExternalStandardDB.java:77) at uk.ac.exeter.QuinCe.data.Instrument.Calibration.CalibrationDB.getTargets(CalibrationDB.java:155) at uk.ac.exeter.QuinCe.data.Instrument.Calibration.CalibrationDB.getCurrentCalibrations(CalibrationDB.java:104) at uk.ac.exeter.QuinCe.web.Instrument.CalibrationBean.loadCurrentCalibrations(CalibrationBean.java:177) at uk.ac.exeter.QuinCe.web.Instrument.CalibrationBean.start(CalibrationBean.java:68) at 
...

Proposed fix: Hide the button if no deployments exist.

jonasfh, 2018-01-30:
ref #669

squaregoldfish, 2018-01-31:
This was caused by an instrument ID left in the user preferences after that instrument was deleted from the database.

This was fixed as part of #669

closed 2018-01-31 by squaregoldfish