quince-science/QuinCe #639: Run Type not added to file_column table for Shcuster data set

jonasfh, 2018-01-19:
When adding instrument for the Schuster data set (and possibly other), the run type is not added to the file columns. This leads to the ExtractDataSetJob failing: 

> uk.ac.exeter.QuinCe.data.Files.DataFileException: java.lang.ArrayIndexOutOfBoundsException: -1
	at uk.ac.exeter.QuinCe.data.Dataset.DataSetRawData.createRecord(DataSetRawData.java:701)
	at uk.ac.exeter.QuinCe.data.Dataset.DataSetRawData.getNextRecord(DataSetRawData.java:268)
	at uk.ac.exeter.QuinCe.jobs.files.ExtractDataSetJob.execute(ExtractDataSetJob.java:96)
	at uk.ac.exeter.QuinCe.jobs.JobThread.run(JobThread.java:85)
Caused by: java.lang.ArrayIndexOutOfBoundsException: -1
	at java.util.ArrayList.elementData(ArrayList.java:422)
	at java.util.ArrayList.get(ArrayList.java:435)
	at uk.ac.exeter.QuinCe.data.Dataset.NoAverageDataSetRawData.getSelectedTime(NoAverageDataSetRawData.java:176)
	at uk.ac.exeter.QuinCe.data.Dataset.DataSetRawData.createRecord(DataSetRawData.java:656)

squaregoldfish, 2018-01-19:
We will have a short term fix to place the Run Type in the SensorTypes for the instrument as a special step in `NewInstrumentBean`.

We need to revisit this to establish the best way to do it properly.

jonasfh, 2018-01-19:
Create SensorType  factory method with the following settings: 
String name   “Run Type”
boolean required    true
boolean named  false
String requiredGroup   null
String dependsOn null
String dependsQuestion null
boolean many  false
boolean averaged false
boolean postCalibrated false
boolean coreSensor false
boolean usedInCaclulation false
boolean externalStandards false

closed 2018-01-22 by squaregoldfish