quince-science/QuinCe #651: Continuous build

squaregoldfish, 2018-01-22:
Need to be able to checkout, build and deploy to a test server as an automatic step.

Include Javadoc too.

jonasfh, 2018-02-12:
Suggested process for release (Might be a single script):

Make new release number (https://semver.org/)

**Lun locally:**
- Rebase the release onto master
- Make a git tag for the release
- Git tag whateverversion 
- Update a changelog.md with changes for this release
- Update version number in build-gradle with the new version number
- Push to the main repository on github

**On production server:**
- Pull latest changes from master branch on main repository
- .gradlew release 
... which does the following: 
    - Build Javadocs
    - Run unit tests
    - Run war - task (with dependencies)
    - Run pre-relase - task, scripts in script/prod/pre-release (where name between version.txt and current version)
    - Run flywayMigrate task, does db migration. Migration files should use version number
    - Run post_release taks, similar to pre-release 
    - Print new version number to version.txt
    - Apply prod configuration: setup_reverse_strings, then  setup_replace_strings.sh
    - Copy task to copy war to tomcat deploy folder
    - Release is aborted if any of the tasks fail!
    - Tomcat automatically reloads new war?!

squaregoldfish, 2019-01-17:
This will probably come in with moving the test server to a Docker framework once the Carbon Portal has figured out how to do it.

squaregoldfish, 2019-12-04:
https://jenkins.io/

squaregoldfish, 2020-07-12:
Build and junit tests are being run as Github actions

squaregoldfish, 2024-07-05:
I'm happy with the setup we have for the time being. Any adjustments will have their own issues.

closed 2024-07-05 by squaregoldfish