quince-science/QuinCe #928: Confirmation email failed in release 1.0.0

jonasfh, 2018-08-14:
When registering a new account in the new QuinCe version v1.0.0, the registration email failed with the following message: 

    java.lang.NumberFormatException: For input string: ''
        at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65)
        at java.lang.Integer.parseInt(Integer.java:592)
        at java.lang.Integer.parseInt(Integer.java:615)
        at uk.ac.exeter.QuinCe.utils.EmailSender.sendEmail(EmailSender.java:63)
        at uk.ac.exeter.QuinCe.jobs.user.SendEmailVerificationMailJob.execute(SendEmailVerificationMailJob.java:58)
        at uk.ac.exeter.QuinCe.jobs.JobThread.run(JobThread.java:79)

jonasfh, 2018-08-14:
Trying to re-queue the job leads to further exceptions: 

java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at uk.ac.exeter.QuinCe.jobs.JobManager.getJobFromResultSet(JobManager.java:483)
	at uk.ac.exeter.QuinCe.jobs.JobManager.getNextJob(JobManager.java:688)
	at uk.ac.exeter.QuinCe.jobs.JobManager.startNextJob(JobManager.java:896)
	at uk.ac.exeter.QuinCe.jobs.JobManager.requeueJobs(JobManager.java:1145)
	at uk.ac.exeter.QuinCe.jobs.JobManager.requeueJobs(JobManager.java:1167)
	at uk.ac.exeter.QuinCe.jobs.JobManager.requeueJob(JobManager.java:1200)
	at uk.ac.exeter.QuinCe.web.jobs.JobsBean.requeue(JobsBean.java:174)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.apache.el.parser.AstValue.invoke(AstValue.java:247)
	at org.apache.el.MethodExpressionImpl.invoke(MethodExpressionImpl.java:267)
	at org.apache.myfaces.view.facelets.el.ContextAwareTagMethodExpression.invoke(ContextAwareTagMethodExpression.java:96)
	at org.apache.myfaces.application.ActionListenerImpl.processAction(ActionListenerImpl.java:74)
	at javax.faces.component.UICommand.broadcast(UICommand.java:120)
	at javax.faces.component.UIViewRoot._broadcastAll(UIViewRoot.java:1172)
	at javax.faces.component.UIViewRoot.broadcastEvents(UIViewRoot.java:365)
	at javax.faces.component.UIViewRoot._process(UIViewRoot.java:1658)
	at javax.faces.component.UIViewRoot.processApplication(UIViewRoot.java:862)
	at org.apache.myfaces.lifecycle.InvokeApplicationExecutor.execute(InvokeApplicationExecutor.java:42)
	at org.apache.myfaces.lifecycle.LifecycleImpl.executePhase(LifecycleImpl.java:196)
	at org.apache.myfaces.lifecycle.LifecycleImpl.execute(LifecycleImpl.java:143)
	at javax.faces.webapp.FacesServlet.service(FacesServlet.java:198)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:292)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:207)
	at org.apache.tomcat.websocket.server.WsFilter.doFilter(WsFilter.java:52)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:240)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:207)
	at uk.ac.exeter.QuinCe.web.User.AuthenticatedFilter.doFilter(AuthenticatedFilter.java:85)
	at org.apache.catalina.core.ApplicationFilterChain.internalDoFilter(ApplicationFilterChain.java:240)
	at org.apache.catalina.core.ApplicationFilterChain.doFilter(ApplicationFilterChain.java:207)
	at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:212)
	at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:106)
	at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:502)
	at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:141)
	at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:79)
	at org.apache.catalina.valves.AbstractAccessLogValve.invoke(AbstractAccessLogValve.java:616)
	at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:88)
	at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:528)
	at org.apache.coyote.http11.AbstractHttp11Processor.process(AbstractHttp11Processor.java:1099)
	at org.apache.coyote.AbstractProtocol$AbstractConnectionHandler.process(AbstractProtocol.java:670)
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1520)
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.run(NioEndpoint.java:1476)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.lang.Thread.run(Thread.java:748)
Caused by: uk.ac.exeter.QuinCe.jobs.InvalidJobParametersException: Unhandled exception while validating parameters
	at uk.ac.exeter.QuinCe.jobs.user.SendEmailVerificationMailJob.validateParameters(SendEmailVerificationMailJob.java:84)
	at uk.ac.exeter.QuinCe.jobs.Job.<init>(Job.java:118)
	at uk.ac.exeter.QuinCe.jobs.user.SendEmailVerificationMailJob.<init>(SendEmailVerificationMailJob.java:46)
	... 52 more
Caused by: uk.ac.exeter.QuinCe.jobs.InvalidJobParametersException: The specified user doesn't have an email verification code
	at uk.ac.exeter.QuinCe.jobs.user.SendEmailVerificationMailJob.validateParameters(SendEmailVerificationMailJob.java:81)
	... 54 more

squaregoldfish, 2018-08-14:
First guess is a configuration issue - is the email port number empty? If that is the problem, it should fall back to using the default.

squaregoldfish, 2018-08-15:
It is something to do with the email port, but it's in the configuration so who knows?

squaregoldfish, 2018-09-25:
I think we need to add the email server properties to the `quince.setup` file and have them processed by `setup_replace_strings.sh`.

closed 2018-10-05 by squaregoldfish