quince-science/QuinCe #1808: More detailed export reporting

squaregoldfish, 2020-08-13:
The export script should provide more detailed reports to help with flagging up issues that will be of interest to us and downstream projects.

If no data is exported, a message to this effect should be sent. This should prompt us to look into the reason.

The script should record when it last successfully exported data to a given destination. Then for the next export it must calculate the time of the first measurement after that point. The difference between that measurement and the current time is the longest delay in processing.

After that, we can generate alerts based on the requirements of each project. At the moment there is only a requirement for CMEMS: if the first measurement is processed before 12:00 UTC on the day after it was taken, then there is no delay to report. Otherwise we must report the number of minutes after that midday point until the data has been uploaded to CMEMS (i.e. the DNT update is accepted).

MarenKK, 2020-09-25:
Add pip-env support

MarenKK, 2021-02-22:
**Missing timer**  
Calculate CMEMS delay