quince-science/QuinCe #2100: Huge negative xCO2 data points not detected by QuinCe range check

cla023, 2021-05-06:
The Finnmaid Los Gatos labelling dataset (34FM20200815) contained about 1500 data points between the range -734199.463 to -6387.644. These were all flagged by QuinCe as good (2). Why were these not detected by the range checker?

[image]

squaregoldfish, 2021-05-11:
The original xCO2 value is fine, so something's gone haywire in during the calibration.

#732 would detect this.

squaregoldfish, 2021-05-27:
The original cause of this was a misreading of missing values (999999), which were processed as valid values. Since this has been fixed in #1873 I'm closing this issue. The referenced issues above will get sorted out in their own time.

closed 2021-05-27 by squaregoldfish