quince-science/QuinCe #2553: CONTROS algorithm throws SQL exception

squaregoldfish, 2022-10-12:
Find and destroy

glncstll, 2022-11-10:
it seems that the query you ask for the mysql database is wrong, you ask for:
SELECT date, value FROM sensor_values WHERE dataset_id = 6 AND file_column IN () ORDER BY date ASC;
and it is wrong
but if you ask for:
SELECT date, value FROM sensor_values WHERE dataset_id = 6 AND file_column ORDER BY date ASC;
without IN () after file_column in the query string it works
also in the error message is written there is error in the sql syntax:
[uk.ac](http://uk.ac/).exeter.QuinCe.utils.DatabaseException: Error while getting run type periods
	[uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1152)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
Caused by: java.sql.SQLSyntaxErrorException: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ‘) ORDER BY date ASC’ at line 1
	com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:120)
	com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:122)
	com.mysql.cj.jdbc.ClientPreparedStatement.executeInternal(ClientPreparedStatement.java:916)
	com.mysql.cj.jdbc.ClientPreparedStatement.executeQuery(ClientPreparedStatement.java:972)
	jdk.internal.reflect.GeneratedMethodAccessor53.invoke(Unknown Source)
	java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	java.base/java.lang.reflect.Method.invoke(Method.java:568)
	org.apache.tomcat.jdbc.pool.StatementFacade$StatementProxy.invoke(StatementFacade.java:118)
	jdk.proxy2/jdk.proxy2.$Proxy15.executeQuery(Unknown Source)
	[uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1145)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobFailedException: Job ID 11 failed
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:294)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
Caused by: [uk.ac](http://uk.ac/).exeter.QuinCe.utils.DatabaseException: Error while getting run type periods
	[uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1152)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobFailedException: Job ID 11 failed
	at [uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:294)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
Caused by: [uk.ac](http://uk.ac/).exeter.QuinCe.utils.DatabaseException: Error while getting run type periods
	at [uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1152)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	... 1 more
Caused by: java.sql.SQLSyntaxErrorException: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ‘) ORDER BY date ASC’ at line 1
	at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:120)
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:122)
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeInternal(ClientPreparedStatement.java:916)
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeQuery(ClientPreparedStatement.java:972)
	at jdk.internal.reflect.GeneratedMethodAccessor53.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.apache.tomcat.jdbc.pool.StatementFacade$StatementProxy.invoke(StatementFacade.java:118)
	at jdk.proxy2/jdk.proxy2.$Proxy15.executeQuery(Unknown Source)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1145)
	... 2 more
[uk.ac](http://uk.ac/).exeter.QuinCe.utils.DatabaseException: Error while getting run type periods
	[uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1152)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	[uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
Caused by: java.sql.SQLSyntaxErrorException: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ‘) ORDER BY date ASC’ at line 1
	com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:120)
	com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:122)
	com.mysql.cj.jdbc.ClientPreparedStatement.executeInternal(ClientPreparedStatement.java:916)
	com.mysql.cj.jdbc.ClientPreparedStatement.executeQuery(ClientPreparedStatement.java:972)
	jdk.internal.reflect.GeneratedMethodAccessor53.invoke(Unknown Source)
	java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[... cut ...]
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.apache.tomcat.jdbc.pool.StatementFacade$StatementProxy.invoke(StatementFacade.java:118)
	at jdk.proxy2/jdk.proxy2.$Proxy15.executeQuery(Unknown Source)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1145)
	... 2 more

the error is caused in the DataSetDataDB.java file see where I have put breakpoints

[image]

squaregoldfish, 2022-12-01:
There's an error during Auto QC that doesn't prevent the dataset being processed, but should nevertheless be eliminated:

```
Sensor Quality Control - error: Error while getting run type periods
[uk.ac](http://uk.ac/).exeter.QuinCe.utils.DatabaseException: Error while getting run type periods
	at [uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1152)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.jobs.files.AutoQCJob.execute(AutoQCJob.java:166)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.jobs.JobThread.run(JobThread.java:89)
Caused by: java.sql.SQLSyntaxErrorException: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ‘) ORDER BY date ASC’ at line 1
	at com.mysql.cj.jdbc.exceptions.SQLError.createSQLException(SQLError.java:120)
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:122)
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeInternal(ClientPreparedStatement.java:916)
	at com.mysql.cj.jdbc.ClientPreparedStatement.executeQuery(ClientPreparedStatement.java:972)
	at jdk.internal.reflect.GeneratedMethodAccessor81.invoke(Unknown Source)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:568)
	at org.apache.tomcat.jdbc.pool.StatementFacade$StatementProxy.invoke(StatementFacade.java:118)
	at jdk.proxy2/jdk.proxy2.$Proxy15.executeQuery(Unknown Source)
	at [uk.ac](http://uk.ac/).exeter.QuinCe.data.Dataset.DataSetDataDB.getRunTypePeriods(DataSetDataDB.java:1145)
	... 2 more
```

squaregoldfish, 2023-08-03:
For datasets without post-calibration, there's a few things to note:

- NRT is OK - it's all "questionable" anyway
- Should not be allowed to Submit
- Exported files need something in manifest and data
  - Tobias suggests Argo notation of R = realtime (no QC) A = automatic QC and D = Delayed mode (L2)
  - Alternatively (currently preferred) use 1 as Not Calibrated (https://exchange-format.readthedocs.io/en/latest/quality.html)

squaregoldfish, 2023-12-30:
Flag order (most to least)

- Lookup (-200)
- Flushing (-100)
- Needed (-10)
- Bad (4)
- Questionable(3)
- **NEW** Not Calibrated (1)
- Good (2) / Assumed Good (-2)
- No QC (0)

squaregoldfish, 2024-01-24:
There's an update to the CONTROS algorithm on the way that uses more columns from the data files.

squaregoldfish, 2024-02-02:
OK, so I did the SST correction stuff wrong.

New sensors can have a T_membrane, which we should use to correct to SST if it's in the data files. So we need to add a question to the CONTROS variable of "Do you have T_membrane sensor?" and then apply that if the user says yes.

squaregoldfish, 2024-04-03:
We're going to publish what we have, and update the algorithm in future if needed.

squaregoldfish, 2024-04-05:
The new algorithm is delayed. We will release with the current algorithm, and any updates will be handled as separate issues.

closed 2024-04-05 by squaregoldfish