quince-science/QuinCe #2583: Investigate an error when upload a Contros data file

glncstll, 2022-11-09:
[image]

glncstll, 2022-11-09:
Problem should be perhaps solved: when I uploaded the instrument in the drag and drop operation I put the date as YYYY-MM-DD but in the file was DD-MM-YYYY 
[image]

squaregoldfish, 2022-11-10:
So the error thrown when the date can't be parsed isn't very helpful. We should make it better.

squaregoldfish, 2025-02-20:
This is less of an issue now that #452 has been finished. However, it can still happen if dd/mm/yy and mm/dd/yy are mixed up.