quince-science/QuinCe #2821: NRT Preprocessor script to fix newlines

squaregoldfish, 2024-01-05:
In GO files, we sometimes see newlines happening in rows where they shouldn't happen. The data is complete - we just get a line with the values that should have been on the previous line. We should be able to write an NRT preprocessor script to fix this.

In the simplest case we just remove the newline - this can be detected because the first field isn't a valid GO Run Type. BUT we must make sure that a reconstructed line has the correct number of columns, otherwise other glitches will go unnoticed (or break the processing in other unpredictable ways).

Also think about whether the GO-specific solution above can be generalised through some kind of configuration (e.g. allowed-first-values or similar)

squaregoldfish, 2024-01-05:
Maybe:

If line is short, check next line. If next line + this line = full line, join them

JoMaHoPo, 2024-01-24:
In the new code I have generated the scrip does as follows:

Opens the file line by line considering the first line to be the header and counts the number of elements in that.

Then:
1) It reads the 'data' lines and counts the number of tabs in that line. 
2) It reads the following line and sums the number of non tab elements. 
3) If the number of non tab elements is larger than the number of columns, it continues.
4) If the number of non tab elements resulting of next line + this line = full line, then it adds the elements of the next line at the end of this line. Deleting the next line. 
5) Continues to a new line and so on.
6) When it reaches the end of the file it writes a new .txt and saves it to a pre defined path.

squaregoldfish, 2024-01-29:
Configuration:

- Seperator
- No of columns? Optional - if not specified, use first line.

JoMaHoPo, 2024-01-29:
The number of columns is not specified but extracted from the first line, which is named header in the code. The separator is established as tab, however I can make it more flexible and look for any separator?

squaregoldfish, 2024-01-29:
Don't worry, they're notes for me when integrating the script into the main NRT processing script

PR #2861 (2024-02-06): Newline fix preprocessor
Fix #2821
closed 2024-02-06 by squaregoldfish